supermerill/SuperSlicer #4576: Support blocker causes crash

tavdog, 2025-04-04:
### What happened?

The program is crashing when painting support blockers.

### Project file & How to reproduce

Sliced the attached 3mf project file.

[Tron Capsule Thicker with curves - Top.3mf.zip](https://github.com/user-attachments/files/19600352/Tron.Capsule.Thicker.with.curves.-.Top.3mf.zip)

### Version

2.7.61

### Operating system

macos 15.3.2

### Printer model

Ender3

supermerill, 2025-04-04:
I opened your project with 2.7.61.0 (on windows, sadly).
then I clicked on the paint icon at the left bar.
then I was that there is already some support blocker.
i paint again (blockers), slice, paint again (blocker and enforcers, slice.
can't make it crash.

I'll try to find another person with a mac to test it out.

tavdog, 2025-04-04:
This is M1 mac.  Yes it only crashes when trying to slice with those painted on support blockers.