supermerill/SuperSlicer #4579: physical printer

jeanmarce, 2025-04-05:
Hello, in version SuperSlicer_2.7.61.0_win64_250325 I can't display my physical printer, even though I'm connected and I can send my STL to my K1max printer.
Thank you for your work. I've always loved your slicer.

supermerill, 2025-04-14:
I need more information.
I took my latest beta release, created a physical printer of a rrandom marlin printer profile I have, with octoprint.
I can see it in the list 

[image]

that's your issue? it's not in the list?

jeanmarce, 2025-04-21:
Hello, I'm using a K1Max networked in SuperSlicer. There's no physical printer tab to view my Klipper settings on my printer like in PrusaLicer. However, I can connect and send my STL files to my printer with SuperSlicer. I hope my explanations are clear.
Have a nice day.

image.png (view on web)<https://github.com/user-attachments/assets/e7d1fc89-4e6a-4f01-8130-4c214f1fdf17>

image.png (view on web)<https://github.com/user-attachments/assets/e7d1fc89-4e6a-4f01-8130-4c214f1fdf17>