supermerill/SuperSlicer #4580: Unwanted gap fill

Tupson444, 2025-04-06:
### What happened?

I'm trying out the new version, and I noticed some gap fill added for no reason:

[image]

in the same project file can be seen some other bugs as well: overhang perimeter where it is vertical (embossed text) and missing parts of the bottom of the text:

[image]

### Project file & How to reproduce

[strange gap fill.3mf.zip](https://github.com/user-attachments/files/19619208/strange.gap.fill.3mf.zip)

### Version

2.7.61.0

### Operating system

win 10

jbeima, 2025-04-10:
Could you please test this with the Beta that was just released on 2025-04-07.

Tupson444, 2025-04-10:
I tried the beta now, but the unwanted gap fill is still there.

But I noticed another bug (perhaps I should open another issue for it?), some lines in the top infill are missing, but it is only shown in the Gcode preview:
[image]

In Sliced preview it looks normal:
[image]

In an online gcode viewer, the same is shown, but on even more places:
[image]

I currently don't have time to print the file to see if it shows in real part.

supermerill, 2025-04-14:
Gapfill :change solid infill pattern from `rectilinear (filled)` to `rectilinear`
precision issue: set the printer-> general -> firmware -> `gcode_min_length` to 0

supermerill, 2025-04-14:
I guess I should not print gapfill it it's not against another solid infill or perimeter, but currently, the detection isn't made.
For the gcode precision: I wanted to revamp this algorithm, and i was worse than the current one. Still have many improvement to not deleted useful points.

Tupson444, 2025-04-14:
Thank you for looking into this.

Regarding gap fill, changing to rectilinear solves the issue (I didn't notice it was set to `rectilinear (filled)`, I should have tried that setting).

For gcode precision, I've turned off max_gcode_per_second (or set to a higher value) and it's back to normal. Somehow I missed it when troubleshooting so I didn't identify it as the cause. Otherwise, I think it is a promising feature to help with old printers that have slow processors to print faster, but I can't see a reduction in gcode commands when setting a low value. I've opened a new issue for it: #4590.