supermerill/SuperSlicer #4581: MacOS transparent top nav bar

rtm7777, 2025-04-07:
### What happened?

MacOS build have transparent background on top bar

[image]

### Project file & How to reproduce

---

### Version

2.7.61.1

### Operating system

macos 15.3.2, arm64

ipa64, 2025-04-17:
I also see this behavior, but in fact it is not the navigation bar that is transparent, it is the parameter list that scrolls over it.

[image]

ipa64, 2025-04-22:
Yes it is not a big issue, but really annoying, each time you navigate in your settings, you have to scroll down the list to get a clear access to icons save, add, search,… otherwise it’s messy.