supermerill/SuperSlicer #4583: Bottom fill pattern setting ignored when using raft

toptensoftware, 2025-04-10:
### What happened?

I'm trying to print a small object with circular bottom that needs a raft.   I want a circular fill pattern on the bottom surface, but this setting only seems to work without the raft.  As soon as I enable the raft, it switches back to rectilinear.

See [video](https://www.dropbox.com/scl/fi/1i3jg1cw26temgnsrbyh9/SuperSlicerRaftBottomInfill.mkv?rlkey=ouo1a44qj6s09p1whqgmsfaw5&dl=0)

This problem doesn't happen in PrusaSlicer or Orca and wondering if I've missed a setting somewhere?
 
Have tested with v2.5.59.13 and 2.7.61.1 - same problem in both.

### Project file & How to reproduce

[Test(SS).zip](https://github.com/user-attachments/files/19675958/Test.SS.zip)

### Version

v2.5.59.13 and 2.7.61.1

### Operating system

Windows 11

### Printer model

Ender 3 V2 Neo

neophyl, 2025-04-10:
Looking at your project, its treating the layer above the raft as internal bridge infill for some reason.  So yeah it doesn't matter what your bottom infill settings are set to its not bottom infill.  Looks like a bug to me.

supermerill, 2025-04-14:
I see that with prusaslicer, there is no overhangs or bridges over the raft, while there is overhangs & bridge over support.
I'll see where is the issue. 

I may add a setting to enable overhangs/bridges over raft.