supermerill/SuperSlicer #4585: Performance issue - any advice?

CuredPrusa, 2025-04-10:
I know most of developers' focus is on 2.7 now. But there are still some of us using 2.5 extensively for bullet proof projects before 2.7.
For this particular project I use nightly from Dec2024

SuperSlicer
Version:   2.5.60.1
Build:     SuperSlicer_2.5.60.1

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 34,049MB
OpenGL installation
GL version:   4.6.0 - Build 32.0.101.6734
Vendor:       Intel
Renderer:     Intel(R) Iris(R) Xe Graphics
GLSL version: 4.60 - Build 32.0.101.6734

I struggle with my PC performance and I see there's something what bothers my Super Slicer. I can't see what that is. I'd appreciate anyone's idea . 

Ok, my project is hefty...

[image]

[image]

[image]

...but still...
Three minutes standing at

[image]
after slicing command, for any small change.

CPU looks like this then:

[image]

After approx 3 minutes of max CPU cooling fan, SS goes to

[image]

Then CPU calms down and smooth layer creation begins which finishes the slicing process. 
I take care to run Super Slicer alone on my PC, without other simultaneous processes. Doesn't help. 

However, sometimes (I can't say when and why!) it jumps over this dreaded "discover shells" step in a second, and slicing is completed in 15 seconds! 

For comparison, I tried to simulate similar project in Orca slicer and it slices much faster. But I don't care for Orca, it's missing a lot of functionality which Super Slicer provides.

legend069, 2025-04-10:
why are you using the height range modifier like that lol?
it works as a range
so start = 0.2
end = 5
will make the layers in the range ``x`` layer height

legend069, 2025-04-10:
like this one 
can be combined into ``start= 0.3 end = 6.9``
the second from the bottom actually goes backwards too.

[image]

if you can repeat this in 59.13(i don't have 2.5.60.1 installed) i can take a look and find out why it takes so long on the shell stage.
theoretically the longest part should always be ``generating gcode.``

we can only make it "faster" by multi-threading more steps. and doing this opens a whole can of worms with "race conditions"

jbeima, 2025-04-10:
This maybe a strange question, but did you try 2.5.59.13? This is out last "stable" version.

Version 2.5.60.0 was a beta and contained several known bugs. Right now we have completely deprecated it.

I can appreciate your wanting to use tried and tested before upgrading, since the newer 2.7 echo system introduces some new methodologies.

Are you on our Discord?

CuredPrusa, 2025-04-10:
Thanks man!
Yes I did try 2.5.59.13. In fact, I came from there last year, with the same project. It seems to me .60 works faster with my settings.  

It seems I am on a Discord, but I use it only when someone tell me to "go there"  :)

At the moment I am trying a fresh iteration of the same version. I'm using a console run. 
It seems a bit better. 

It all went south when I added just a bit more of range and object modifiers. It looks like I've crossed "the edge" of "just enough". :)

off topic
Btw, I have noticed that "if-elsif" macro has a limitation of variable count. When I enter more than 3, it goes bananas. This is why I have to repeat the same layer range with elseif, while I could actually put more in just one "if" line. 

Like here:

[image]
if you disregard M221, I could put solidinfill and gapinfil together (combined with "and") but I can't. I figured out the hard way -3 variables are the max.

CuredPrusa, 2025-04-10:
After PC reboot, it became better.
I've tested now both .59 and .60.
Both begin layer slicing after 42 seconds. So in a bit below minute, slicing is done.
But I will stick to .60 because the layer presentation with gcode is accurate, unlike .59. And some other improvements.

CuredPrusa, 2025-04-10:
> Version 2.5.60.0 was a beta and contained several known bugs. 

Could you tell on which details should a pay attention, considering those bugs?

legend069, 2025-04-11:
> I was just hoping there's some trick to boost it up, something ridiculous as "have it all in folder close to root" - a tip I remember from Marlin firmware with Visual Studio Code.

not that i know of.
since you're on a laptop there might be some weird power savings related things windows is doing.

CuredPrusa, 2025-04-11:
Thank you all for contribution.
Here a fresh update.
Whole day today I was slicing this project within 15 seconds! 
But what bothers me now.... why? I don't know what I've done, and why did it act so slow before. For quite a long period, for months. Of course, with this particular project. Smaller projects went in a flash, as always. 
Maybe I have changed some modifier which triggered a headache. Yes, I change them on daily basis for fine fine tuning. 
Will this better performance last? I don't know.

CuredPrusa, 2025-04-19:
So, just to confirm... Sometimes it can hang here for several minutes:

[image]

While sometimes I can not even see that step because it goes immediately to smooth layer slicing and finishes within 15 seconds: 

[image]

The same project, just some minor changes here and there. I still can't figure out what changes make it slow.

supermerill, 2025-04-21:
Can't say for sure without the project file to replicate it and using the profiler.
It's mostly when there is many disconnected areas on the same layers, with a O(n3) algorithm, can hang up the slicer.
I don't know where to optimise without a problematic project to investigate.

CuredPrusa, 2025-04-22:
Understandable, Thank you anyway!  If you have in mind sections where there are more separate external perimeters in the same layer, yes that is true, there are 4 in average, max 6. 

For what is worth, 2.5.59.13 suffers the same, but I'm back to it due to some bugs I have found in newer versions of 2.5. 
After a hint from [jbeima](https://github.com/jbeima) I carefully examined some critical points and noticed unexpected retractions where line end shouldn't be retracted. For example, transition from external perimeter to internal. Having it inside the model (functional engineering device) I didn't notice that disturbance which could actually jeopardise structural and functional integrity of the printed model. Therefore I will not continue my adventures lurking away from old and proven  2.5.59.13, as [jbeima](https://github.com/jbeima) suggested. 

I don't want to share my project, but I neither want to mystify it. It's a special device which should not only be waterproof but should withstand 5bar of water pressure and air pressure delivered into it. Until you try, you can't imagine how hard it is to do it on FDM printer within reasonable print time and material consumption.

legend069, 2025-04-22:
> I don't want to share my project, but I neither want to mystify it. It's a special device which should not only be waterproof but should withstand 5bar of water pressure and air pressure delivered [...]

If you can recreate the issue with a standard cube/shape that should be enough for us to investigate.

But it's more then likely it's the model that is causing the issues, maybe theres another model with similar geometry?

CuredPrusa, 2025-04-22:
Yes, it's certainly the model, not so much the settings, although overwhelming. Cube is done in 5 seconds. 
There's nothing similar around. Not many crazy people like myself either.  
Thanks anyway.

jbeima, 2025-04-30:
If it helps, we can take the model submission offline and treat it under an NDA. You would not be the first to have sensitive use case situations.

The reason I am asking, is if we can spend some time with it in the 2.7.6[1|2] code base, maybe we might be able to improve our next stable cases for these rare cases.

I was going to ask if Windows or some background app be running that happened to be shut down the day you had the 15 second runs? We all know how WinTendo can be at times. :)

CuredPrusa, 2025-04-30:
Thanks a lot. I'll think about it.
Now the situation got a bit better after I have made some modifier optimisation. Or was it model optimisation done in parallel, I don't know. 
Slicing is rendered fast when I don't touch layer structure. Tweaking seams, temp, fan, speed is not bad at all. 
10 seconds, all in Generating G-code step. 
As soon as I touch width, pericount, infildens ....  it dives into deep blue. But not too much, 45sec! Doesn't go to disphotic zone as before (3 minutes). Then I take my guitar which sits by my side and start practicing. I became not a bad guitar player with this project. 

A tip for those struggling the same:
Example: Adjusting the solid infill width to get 100% coverage, especially with concentric (or to avoid overlapping lines with monotonic), can be a tedious task due to long waiting for each try, especially when surface is complex enough, so finding the right width on one side messes up the arrangement on the other.  
Workaround: Make a copy of the project. In that copy cut away all irrelevant chunks of the model. That would speed up slicing and your tweaking process. When you find the correct number, just copy it to the original project.

legend069, 2025-05-28:
> I'm back to 4 minutes waiting in this step:
> 
> ![Image] 
> As soon as I change something related to infill geometry or perimeters geometry I can go to kitchen, do some dishes and make me a coffee.
> 
> Is there something hardware-wise that can be enhanced to boost it? However, looking at processor load, it doesn't seem too much. It rather looks like SS struggles within itself.

can you please share the project file so we can look into it.
processor load will only go up if we implement ``multi-threading`` for every step. doing this will cause alot of issues, so it's not currently in our priority list