supermerill/SuperSlicer #4586: Simple fix for 'The application "Finder/Dock" does not have permission to open "(null)."'

ryanknutson, 2025-04-11:
### What happened?

Got this error and did some digging, turns out the fix is simple!

Just delete the following from /Applications/SuperSlicer.app/Contents/Info.plist:
```
  <key>LSEnvironment</key>
  <dict>
    <key>ASAN_OPTIONS</key>
    <string>detect_container_overflow=0</string>
  </dict>
```

Thanks to @bastimeyer [here](https://github.com/streamlink/streamlink-twitch-gui/issues/1027#issuecomment-2449789501)!

### Project file & How to reproduce

N/A

### Version

2.7-beta

### Operating system

macOS 15

### Printer model

N/A

supermerill, 2025-04-14:
thanks, deleted for next version

sala91, 2025-04-28:
does not fix it for me, nor is it fixed with latest beta release

julian-weinert, 2025-11-18:
Same here. I can't open SuperSlicer at all on my mac with the latest shit OS installed. Unfortunately SuperSlicer  decided to not support macOS 10.14 anymore, so I can't use my actual computer that still has a functioning OS on it