supermerill/SuperSlicer #4587: Weird arc travels at seams

sneakypurplegit, 2025-04-12:
### What happened?

New (2.7) built-in arc fitter outputs extra travel moves at most seams on round surface. Both ArcWelder and Bambu options in printer settings. Old external AcrWelder script works fine. 
Extra arcs are after arc perimeter extrusion, before skip to next perimeter. Some are tiny and harmless, some go full circle, far outside of the model.  Size of seam notch doesn't matter. At "Seam position: Rear" extra moves are still there, but are all tiny. 

### Project file & How to reproduce

### Version

2.7.61.1

### Operating system

Ubuntu 24.04

### Printer model

Marlin 2 @ RAMPS 1.4

supermerill, 2025-04-14:
Can you send me a project file instead of .ini?

sneakypurplegit, 2025-04-14:
Here. Project (default cylinder, zero infill for haste), gcode, and printer .ini, with enabled new built-in arc fitting. The weird moves are travels, hidden in viewer at first.

[Shape-Cylinder.3mf.zip](https://github.com/user-attachments/files/19742639/Shape-Cylinder.3mf.zip)

Also I tried Windows build (via Wine) with my configs, same result there.