supermerill/SuperSlicer #4588: first_layer_extrusion_spacing GUI spacing is not calculated correctly ?

legend069, 2025-04-13:
### What happened?

perimeter_overlap and external_perimeter_overlap both effect the first_layer_width but the spacing field doesn't get updated to reflect this, is that a bug or expected behavior ?

[image]

### Project file & How to reproduce

```
first_layer_flow = Flow::new_from_config(FlowRole::frPerimeter, *print_config, nozzle_diameter, first_layer_height, 1.f, true);
first_layer_spacing = first_layer_flow.spacing();
```

```double first_layer_spacing = full_print_config.get_abs_value("first_layer_extrusion_spacing", nozzle_diameter);```

supermerill, 2025-04-14:
the first_layer_extrusion_spacing/first_layer_extrusion_width don't uses the perimeter overlap.
It's currently using the min between 1 and filament_max_overlap
that's because it can be used for everything, like wipetower, skirt, brim, ...  (and infill if no first_layer_infill_extrusion_width

I didn't though about that when creating the first_layer_infill_extrusion_width, so I guess I may use the solid_infill_overlap, as it should mostly use this (sparse & top infill on first layer are rare things)

legend069, 2025-04-14:
changing the overlap does effect the gcode for the first layer so the spacing would be changing too.
unless it's not getting applied here?

first layer with perimeter_overlap and external_perimeter_overlap set to 0%
[image]

first layer with perimeter_overlap and external_perimeter_overlap set to 80%
[image]

supermerill, 2025-04-14:
Oh, ok. More smart than I though.
Still, what to display? the perimter one? the solid infil one? the externalperimeter one?
I choose to let it at the filament one, as I can't see a way.

supermerill, 2025-04-15:
> it currently uses a combo of perimeter and external, the GUI should reflect that ?

I think only for the external perimeter of the first layer.

I think i'll just add a note in the tooltip.