supermerill/SuperSlicer #4595: M104 vs M109 Causing Print Start Delay

SkiBikePrint, 2025-04-16:
I started testing v2.7.61.1 today versus the 2.5.59.13 I've been running.  Thanks for the organic tree supports btw!

Maybe a relevant note...I am running an ERCF MMU with Happy Hare and my own version of Blobifier for purging so there is a lot going on when a g-code file is generated.  Anyway,

I noticed a variable long pause after MMU initialization, tool load, and filament purge when the standard printing g-code takes over.  Up to 80-90 seconds (sometimes only 10).  Comparing the model sliced with v2.5 vs 2.7 I see that the toolhead temperature command changed from M104 (don't wait for temp) to M109 (wait).  I have never needed to wait since I just got done purging at temperature.  Unfortunately on my setup sometimes it takes a long time for temperatures to settle within range so I believe the pause is due to some temp bouncing around the set point a little and the M109 command forcing the wait.

I believe the M109 vs M104 is is a change from SS and prefer the previous method of M104.

[image]

SkiBikePrint, 2025-04-17:
Nevermind!  I went to the SS Discord and became aware of the "Emit temperature Commands Automatically" check box under printer custom g-code.

legend069, 2025-04-17:
Glad you're able solve it.

For future reference you're more than welcome to ask in our discord before creating a GitHub issue ☺️

SkiBikePrint, 2025-04-17:
Right, of course.  Embarrassingly, I wasn't aware of your Discord until after I entered the issue and then it occurred to me that you must have one.

Sent from my Verizon, Samsung Galaxy smartphone

closed 2025-04-21 by supermerill

SkiBikePrint, 2025-04-21:
Reopening following further testing...

After further testing I am seeing the correct behavior of no additional M109 command for extruder "default" and extruder 1 but if any other extruder is selected the M109 command is inserted before the print is initiated.  I do have an M104 command in my Custom Start G-Code.  Having the extra M109 command creates an unwanted wait, sometimes as long as 90 seconds, before print start even though the hotend has already come up to temp and purged.

SS v2.5.59.13 version does not have the "Emit temperature commands automatically:" option and it automatically inserts an M104 command (don't wait) which is fine.  So perhaps the simplest thing to do is have the new version with the enhanced "Emit temperature Commands Automatically" use M104 vs M109.