supermerill/SuperSlicer #4596: XY clearance for Extruder

ThoriumPrime, 2025-04-20:
This is related to extruder clearance, and solves a problem for those of us who have highly asymmetrical extrusion heads.

1. Currently extrusion clearance only takes into account radius. 
2. I would like to see 4 parameter setting, +x, -x, +y and -y. This would capture a rectangle with the four relevant dimensions from the nozzle. 
3. When these settings are set, the resulting rectangle is evaluated for interference when placing objects. 
4. This would make a big different for us with asymmetric heads, in my case the values range from 25mm to 80mm across different dimensions. (Ratrig vcore 3.1, 200). Hence the 80mm setting eats up half the build plate, even for scenarios that would not necessarily cause interference.

jbeima, 2025-04-20:
Please hop over to our website and submit this via the new Feature Request form, and I will get it on our list of requests.

ThoriumPrime, 2025-04-20:
Unfortunately unable to find a feature request form on superslicer.net...

Calcousin55, 2025-04-21:
The .net one is not operated by the team. But the superslicer.org one is and is the website you are looking for.

ThoriumPrime, 2025-04-21:
Thank you,, I was not aware there was a competing site. The superslicer.net is much more prominent than superslicer.org.

jbeima, 2025-04-30:
This is not a bug it is a Feature Request. :) Which is done from the Feature Request page on the website. (.org)

supermerill, 2025-05-11:
Yes, it needs to be moved from print to printer.
with a narrow clearance for the nozzle part, to know the "narrow clearance" around the nozzle for only some mm.
And then the shape of the extruder, that can be of any shape and maybe not centered.
Also the kind of carriage for the printer:
 * with Y Bar, what's the clearance of the bar?
 * with delta, what's the max z.

Ideally, there should be a "extruder/tool" configuration, and you affect it to the "slot", with a filament.

My main issue right now is that the latest auto-arrange algorithm made by prusa is very complex and it may be hard to modify it to allow any shape. The current offset is very simple to implement.