supermerill/SuperSlicer #4598: "No Perimeters on Bridge Area" -> "Fill the Voids with Bridges" is not working

Kbdeir, 2025-04-23:
### What happened?

I have a model that has a hole in mid air. Using "No Perimeters on Bridge Area" -> "Fill the Voids with Bridges" is expected to generate a sacrificial bridge over the hole. However, actual behavior leaves some portions of the bridge semi-attached and the hole is not covered.

[image].

[image]

Please see attached project file to reproduce the behaviour.

Thanks

### Project file & How to reproduce

[Sacrificial Bridge Bug Report - Superslicer - Voron ABS Profile.zip](https://github.com/user-attachments/files/19864801/Sacrificial.Bridge.Bug.Report.-.Superslicer.-.Voron.ABS.Profile.zip)

### Version

2.7.61.2

### Operating system

Windows 11

### Printer model

Voron 2.4

jbeima, 2025-04-30:
I can re-create this, and will move it to the dev team.

jbeima, 2025-04-30:
With the Linux release, so it is all OS version.

jbeima, 2025-04-30:
We will need to rename this "issue" since it is working, it is just not working as you expect it. It is because there is a hole going through the entire part above it.

Kbdeir, 2025-04-30:
I would say even if there is a hole through the entire part above it, it should still produce a proper sacrificial bridge as under under stable version 2.5.59.13. I have other models which are simply a 5mm nut hole  on a flat 10 mm surface and the sacrificial bridge is not created properly as well. 

[image]

File to reproduce the behavior: 
[HemeraKSBMountLargeCarriage_SeparateBikuHolder.zip](https://github.com/user-attachments/files/19972597/HemeraKSBMountLargeCarriage_SeparateBikuHolder.zip)

Thanks

jbeima, 2025-05-11:
The slicer does not have a "sacrificial bridge" mode, per-say. It was just a happy result of how it was working before.

Now given this, I do agree with you. Since creating a cantilevered bridge here just create challenges of it's own.

I think a new "mode" might be the better option, but that brings up the question, should it be all or nothing, or should there be a way to target it to a specific location?