supermerill/SuperSlicer #4599: On a multiple-monitor setup, superslicer can segfault on splash image scaling

kainz, 2025-04-23:
### What happened?

On a multimonitor setup, the calculations for splash screen scaling can produce negative scaling factors, thus attempting to allocate a splash bitmap of negative size, producing segfaults on use/prefill.

### Project file & How to reproduce

On my 3-monitor setup, GetClientArea as called in `src/slic3r/GUI/GUI_App.cpp` returns `{x = 6400, y = 0, width = 1080, height = 1920}`, producing negative scaling factors.  This scaling code should probably be omitted/reworked.

(commenting out the size vs reported ppi scaling aka lines ~224-245 solved the SIGSEGV on my machine.)

### Version

2.7.61.2

### Operating system

ubuntu 22.04 sway

jbeima, 2025-05-03:
I have passed this to our dev team to see if anyone on it also has a 3 or more monitor setup. I personally have always used SS with dual monitors and have NOT seen this. That being said, this could very well crop up with more than 2.

legend069, 2025-05-04:
what is the orientation of your monitor setup ?
ie;
```
[1][2]
 [3]

[1][2][3]
```

kainz, 2025-05-04:
> what is the orientation of your monitor setup ? ie;
> 
> ```
> [1][2][3]
> ```

Horizonal arrangement, with monitor 3 rotated 90 degress clockwise.

```
    output DP-1 mode 3840x2160@60 position 0,0 adaptive_sync on
    output DP-2 mode 2560x1440@165 position 3840,0 adaptive_sync on
    output HDMI-A-1 transform 90 position 6400,0
```

kainz, 2025-05-13:
i might take a stab at a proper PR to fix this instead of the dummying out of the feature I have in my local branch.  What's are the intended on-screen effects of the splash screen scaling code?

supermerill, 2025-05-26:
> i might take a stab at a proper PR to fix this instead of the dummying out of the feature I have in my local branch. What's are the intended on-screen effects of the splash screen scaling code?

On my 4k laptop monitor, the image was very small, the idea was to scale it a bit so it's up to 75% of the screen.
But I forgot that it may be too big in a 32" monitor, so now I use the ppi to infer what a good size should be.

I think i don't need to use x & y, i'll remove those and add some failsafe.