supermerill/SuperSlicer #4600: Flow::spacing() produced negative spacing for a specific model

kidgenius18, 2025-04-24:
### What happened?

Slicing this model with grid supports turned on results in "Flow:spacing() produced negative spacing for a specific model".  Tried generating grid supports for other models and didn't find any issues.  Organic supports work fine and Prusaslicer added grid supports with the identical model and similar configuration settings.

### Project file & How to reproduce

[Body_Shell.3mf.zip](https://github.com/user-attachments/files/19889750/Body_Shell.3mf.zip)

### Version

2.7.61.2

### Operating system

MacOS Sequoia 15.1.1

### Printer model

Voron 2.4 300mm

kidgenius18, 2025-05-22:
Checked against version 2.7.61.3 and the same issues are present.

supermerill, 2025-05-26:
Your extruder's max layer height should be 75% and not 75.
Maybe it's a problematic convert? It was a profile from prusa?
I'm interested to understand how this field has lost its "%", to avoid it happening again.