supermerill/SuperSlicer #4601: Buttons have no text in "Generate" windows

osgjps, 2025-04-26:
### What happened?

In several of the windows for the various Calibration windows, the buttons at the bottom of the window have no text/have invisible text in them.  Don't know what button to click on for what option.

[image]
[image]

### Project file & How to reproduce

This is an empty project.  Nothing on the platter, but shows configurations.  It's a .3mf file, not CSV but github wouldn't let me upload a 3mf file.

[EmptyProject.csv](https://github.com/user-attachments/files/19923855/EmptyProject.csv)

### Version

2.7.61.2

### Operating system

MacOS 15.3.1

### Printer model

Anycubic Vyper

legend069, 2025-04-27:
This looks like a dark mode related bug

osgjps, 2025-04-28:
Yep.  I switch to light mode and everything looks right.

jbeima, 2025-04-30:
Just an FYI for you. If you zip the 3mf file you can upload it. :) However since this has already been identified as a dark vs light mode issue, we don't need anything. This is just for next time. :)