supermerill/SuperSlicer #4602: Broken UI on Mac (Sequoia, M1 Pro)

Batambatam, 2025-04-30:
### What happened?

After update from 2.4.x version, UI changed a bit (looks like old html pages)
When scrolling on settings pages, UI items overlap the top menu buttons.

[image]

### Project file & How to reproduce

Scroll through settings on any page

### Operating system

macOS 15

### Printer model

voron 2.4

kidgenius18, 2025-05-02:
Believe this is a duplicate of https://github.com/supermerill/SuperSlicer/issues/4256

ipa64, 2025-05-07:
Another duplicate https://github.com/supermerill/SuperSlicer/issues/4581