supermerill/SuperSlicer #4603: Save last used "print settings" and "filament_settings" for each printer

SinisterRj, 2025-05-03:
**Is your feature request related to a problem? Please describe.**
When using multiple printers with Susie, always when I change the printer I need to select the print settings profile and filament profile again. So, in a scenario where I use two printer at the same time to speed up the production, I need to keep switching the configurations every time.

**Describe the solution you'd like**
When I select a printer, it reloads the last profiles used with that specific printer.

**Describe alternatives you've considered**
Another option would be to have different instances of the slicer, one for each printer.

printicus, 2025-05-03:
But you can open two instances at the same time right now....

jbeima, 2025-05-03:
@SinisterRj This is a feature request, please head over to our website ("https://www.superslicer.org"), surf to the Feature Pages, and submit a feature request. I am happy to add this one to the list, since others have asked me about it before.

SinisterRj, 2025-05-10:
> [@SinisterRj](https://github.com/SinisterRj) This is a feature request, please head over to our website ("https://www.superslicer.org"), surf to the Feature Pages, and submit a feature request. I am happy to add this one to the list, since others have asked me about it before.

Yes, it is a feature request, but the link is broken...

SinisterRj, 2025-05-10:
> But you can open two instances at the same time right now....

I'm using it on macOS Arm. I don't know how to do it here.

jbeima, 2025-05-11:
@SinisterRj Link works fine for me and other requests are coming in, so....