supermerill/SuperSlicer #4604: Machine limits transferred when they shouldn't

CuredPrusa, 2025-05-03:
Am I the first one to notice machine limits (acceleration) are transferred to gcode although it wasn't selected?
So when I want to estimate the time I should select "for estimate" and then "disable". 
This is 2.5. Is the new version corrected?
Moreover, why isn't the estimation taken from Speed section where all accelerations are already defined anyway?

[image]

jbeima, 2025-05-03:
Machine limits are always the final say. Since this allows print profiles to be assigned to different machines or machine types. Meaning anything gcode generated will always be capped by machine limits.

Not all firmware support machine limits, for example Klipper does not, which is why I do believe if you select Klipper as your firmware this drop-down can not even be changed.

Some flavours of firmware, support sending max limits separate from request move accels. This would be what this switch should be effecting.

One thing most people forget, is 2.7 is not exactly a replacement or patch to 2.5, it is almost an entire new code base. It is more akin to starting from scratch than patching the old.

SIDE NOTE: This is not the place to ask questions. If you have questions please ask them on Discord. This is for verified bugs.

I believe I can speak on behalf of our team in that we do appreciate what you personally submit, but please also realise that 2.5.60.X is deprecated, and 2.5.59.X will not be having any future work done on it. We don't have the manpower at this point in time to manage two completely separate development branches.

I would take this time to point out, you would be more than welcome to jump into our Discord and please feel free to be more active in the dev channel there.

CuredPrusa, 2025-05-04:
Useful insight. 
I guess I'll visit Discord instead of scribbling here ;) 

Please write me a link to the place you propose, so I don't wander off. 
Thanks

>

closed 2025-05-04 by CuredPrusa

jbeima, 2025-05-04:
We are actually, we have a new docs team that is starting to formally put all this down. Which is why I suggested you come join the daily convo. Come be part of it all.

jbeima, 2025-05-04:
https://discord.gg/4TVUG4VRzP

"In Theory" chats always seem to be well discussed in dev -> dev.