supermerill/SuperSlicer #4605: Current Appimage does not run on my Linux Mint 22.1

bianchifan, 2025-05-04:
### What happened?

System:
[image]

Neither version ubuntu_20.04-2.7.61.2 nor the gtk2 version will start - nothing happens on screen.
They also do not start with help from AppImageLauncher :(
Running from a terminal with activated launcher serves the following output:

[SuperSlicer-ubuntu_20.04-2.7.61.2.AppImage.zip](https://github.com/user-attachments/files/20028385/SuperSlicer-ubuntu_20.04-2.7.61.2.AppImage.zip)

### Project file & How to reproduce

Neither version ubuntu_20.04-2.7.61.2 nor the gtk2 version will start - nothing happens on screen.
They also do not start with help from AppImageLauncher :(
Running from a terminal with activated launcher serves the following output:

[SuperSlicer-ubuntu_20.04-2.7.61.2.AppImage.zip](https://github.com/user-attachments/files/20028385/SuperSlicer-ubuntu_20.04-2.7.61.2.AppImage.zip)

### Version

2.7.61.2

### Operating system

Linux Mint 22.1

jbeima, 2025-05-04:
What happens if you use the Linux binary that is available?

jbeima, 2025-05-04:
Couple of more questions please.

What is the last version that did launch for you?

Can you try moving your "config" directory out of the way, or handing it a "--datadir ./test" to allow it to create a fresh one?

bianchifan, 2025-05-04:
> What happens if you use the Linux binary that is available?

Didn't noticed, the binary will start...thx!

> What is the last version that did launch for you?

It's my first try with Superslicer.
I was out from 3D printing for several years.
Last Slicer i tried were Matterslice, Slic3r and Cura in 2016 IIRC...