supermerill/SuperSlicer #4610: Calibration - On MacOs the buttons doesnt have the text

cristianku, 2025-05-14:
### What happened?

Calibration - On MacOs the buttons doesnt have the text

All the calibration forms, everything is ok, except the buttons doesn't have the text. The buttons works, but without a text its difficult to "imagine" what the button is for

### Project file & How to reproduce

Superslicer, calibration, retraction calibration, scroll down the page ( the form ) , and the button are without text

cristianku, 2025-05-14:
[image]

jbeima, 2025-05-15:
See https://github.com/supermerill/SuperSlicer/issues/4601