supermerill/SuperSlicer #4617: "Conflicts in G-Code paths" when slicing multiple instances with support

dl6lr, 2025-05-23:
### What happened?

Using multiple instances, the arranged parts may be too close arranged. When using support, the first layer touches or overlaps. Upon generation, the warning box "Conflicts in G-Code paths have been detected at layer 0, z=0.00mm." is shown. See the attached project file and the screenshot with the overlapping support when using organic.

[image]

This seems to be no regression, 2.5.59.13 behaves the same, but does not report it. And PrusaSlicer has the same issue.

### Project file & How to reproduce

[upper4x6PLA.zip](https://github.com/user-attachments/files/20409552/upper4x6PLA.zip)

### Version

2.7.61.3

### Operating system

Windows 10

### Printer model

BLV mgn cube

supermerill, 2025-05-25:
Please just use a higher margin for the auto-arrange.
Currently, the auto-arrange can't use the support, and I'm not planning to change that in the near future, as the auto-arrange is already very complex and can't be adjusted for it easily.
Also, the support is computed after the arrange, so it may need to pre-slice before doing the arrange, which may be even more difficult to implement, as it means altering the current workflow.

Dx-Dx-Dx, 2025-07-03:
Or "just" merge arranged parts. 
Why it is not default behavior... 

[image]

supermerill, 2025-07-18:
> Why it is not default behavior...

you can't have object-wide modifiers if you do that.
The merging process may be very complicated.