supermerill/SuperSlicer #4623: Exclude Object markers on MMU Prints

JesNoFear, 2025-05-29:
### What happened?

If you exclude an object on a MMU print, and the object is transitioning between tools (T0 -> T1) Klipper locks xyz movements till the object is ended. thus putting it in a stuck loop.

to validate this, i ran "EXCLUDE_OBJECT_START NAME=BeeLg_stl_id_0_copy_0" "EXCLUDE_OBJECT_END NAME=BeeLg_stl_id_0_copy_0" to unlock xyz and the tool change started right up.

what appears to need to occur is when superslicer is defining object it starts and stops the exclude object, but is not ending and starting the exclude object while on the same object for a tool change. So if you add the end exclude, then trigger the tool change, then start that object again would resolve the issue.

### Project file & How to reproduce

[V350-SS-HQ-MMU-Cute Bee Lg (0,1,3) - ReRun.zip](https://github.com/user-attachments/files/20500632/V350-SS-HQ-MMU-Cute.Bee.Lg.0.1.3.-.ReRun.zip)

running the attached file, if you exclude objects BeeLg_stl_id_0_copy_1 and BeeLg_stl_id_0_copy_7 after the first layer, on Layer 14, when transitioning from tool T0 to T2 klipper will hang in a locked state.

I have a recording of this as well if you would like to enjoy my frustration :-) 

### Version

Version 2.5.59.13

### Operating system

Windows 10

### Printer model

Voron 2.4 R2, ERCF v2, XOL tool head, Voron TAP

JesNoFear, 2025-05-29:
[Cute Bee Lg (0,1,3) - ReRun.zip](https://github.com/user-attachments/files/20500832/Cute.Bee.Lg.0.1.3.-.ReRun.zip)

here is also the 3mf file, it occurred to me that it would be helpful to test against to validate the code fix.

JesNoFear, 2025-05-29:
interesting note as well, Looking on lines 109580, 109592, 109635. This DOES look to be sequenced properly where the termination of EXCLUDE_OBJECT_END NAME=BeeLg_stl_id_0_copy_7 to tool change to start of EXCLUDE_OBJECT_START NAME=BeeLg_stl_id_0_copy_0.
but this seems to be the ONLY occurrence where this takes place.

so maybe it was attempted to be fixed but is not continuing throughout the process?

supermerill, 2025-07-18:
I reworked the "exclude object" boundaries in 2.7, can you try with the latest beta?