supermerill/SuperSlicer #4629: Dithered color mixing idea

VanessaE, 2025-06-11:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**

A color mixing method for printers that only have color-switching, such as Prusa MMU, or machines with multiple extruders feeding a single hotend, or printers with multiple hotends

**Describe how it would work**

So, some of us were talking about color mixing and how existing hobby-level solutions are not very good (such as multi-input hotends), owing to issues such as undesirable laminar flow inside the nozzle.

We brainstormed various ways to make it more accessible, and I proposed the solution of offsetting layers, in a manner similar in concept to dithering in a low-color image (as one does in retro 8-bit graphics).  In a way, this would be like the existing ironing feature, but applied over and over in alternating colors.

It would work like so:

1. For each typical 0.1 to 0.2 mm layer that is to be printed in some blend of the available colors, subdivide that layer into five to ten thinner sub-layers, each being 0.01 to 0.02 mm thick.  Treat these like "coats" of paint.  In fact, I'll just use that term from here on.
2. Perform whatever steps are normally used to change from one filament to another.
3. Print the first/next coat of the layer.  
4. Shift the logical horizontal position of the toolhead in any random direction by 0.01 mm.
5. Change to whatever is the next filament in the set that's being "mixed".
6. Print the next coat of the layer.
7. Return the toolhead's logical position back to what it's "supposed" to be.
8. Repeat steps 2-7 indefinitely, coat after coat, layer after layer, until you reach the next actual color change in the object being printed.

For instance, let's say you need to make green using blue and yellow filaments.  You would print these successive coats in blue and yellow alternately. Similarly, if you wanted to produce a light purple color, and you have blue, red, and white available, you would simply rotate through all three.

You could alter the pattern of color changes to vary the blending ratio of course, and it would work in the same manner as one would chage the duty cycle of a PWM output to vary its effective voltage or current.

Important:  You will need a purge block of course, and you do want to "ram" just before withdrawing the filament to ensure a clean end on it, but you DO NOT want to actually purge the color between coats.  We want to take advantage of the fact that there will be some residue left after each filament swap.

The only time you should actually purge is when changing from one color or color blend to another, and this is just to ensure a sharp transition between the two.

In regards to shifting every other coat, you could also just vary the flow rate back and forth between coats, or you could "expand" the model back and forth (you might be able to exploit the "Slicing: Modifying Slices: XY Compensation" feature for this).  The whole point here is that each coat should have a tendency to "spill over" the edges  of the coat below ever so slightly -- not enough to impact the wall quality, but definitely enough to induce the eye to blend the colors.

The user would need to be able to specify the minimum "coat thickness", the maximum amount of coat-to-coat shift/bloat/whatever.

**Describe alternatives you've considered**

No other alternative exists that lends itself to the hobbyist community.  The closest to this idea would be Polyjet, but that's powder-and-binder tech, not FDM.

**Additional context**

Here's a highly exaggerated drawing of what I'm proposing.  Note that I did not bother to simulate the visual mixing effect:

[image]

jbeima, 2025-06-17:
Please submit this via the "Request A Feature" form from our website.

https://www.superslicer.org

VanessaE, 2025-06-17:
Done.