supermerill/SuperSlicer #4643: SuperSlicer 2.7.61.6 - Crash when importing/opening a step files.

Kbdeir, 2025-07-08:
### What happened?

SuperSlicer crashes when importing a step file.

### Project file & How to reproduce

Open attached project.
Import step file
SS will crash in about 5 seconds.

### Version

2.7.61.6

clarkschaefer, 2025-07-08:
STEP file import is typically a little unstable and notoriously hard to import with software that isn't a professional CAD package, and is not as reliable as STL import because the software has to decide for you how to decimate your smooth surfaces. In this case, since your model has 12 flat faces, you should really just be exporting an STL file from your modeling software and importing that. Interesting that it seems to crash with any STEP that I throw at it.

````

************* Preparing the environment for Debugger Extensions Gallery repositories **************
   ExtensionRepository : Implicit
   UseExperimentalFeatureForNugetShare : false
   AllowNugetExeUpdate : false
   NonInteractiveNuget : true
   AllowNugetMSCredentialProviderInstall : true
   AllowParallelInitializationOfLocalRepositories : true
   EnableRedirectToChakraJsProvider : false

   -- Configuring repositories
      ----> Repository : LocalInstalled, Enabled: true
      ----> Repository : UserExtensions, Enabled: true

>>>>>>>>>>>>> Preparing the environment for Debugger Extensions Gallery repositories completed, duration 0.000 seconds

************* Waiting for Debugger Extensions Gallery to Initialize **************

>>>>>>>>>>>>> Waiting for Debugger Extensions Gallery to Initialize completed, duration 0.031 seconds
   ----> Repository : UserExtensions, Enabled: true, Packages count: 0
   ----> Repository : LocalInstalled, Enabled: true, Packages count: 42

Microsoft (R) Windows Debugger Version 10.0.27704.1001 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File <redacted>
User Mini Dump File: Only registers, stack and portions of memory are available

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*
Symbol search path is: srv*
Executable search path is: 
Windows 10 Version 26100 MP (24 procs) Free x64
Product: WinNt, suite: SingleUserTS
Edition build lab: 26100.1.amd64fre.ge_release.240331-1435
Debug session time: Tue Jul  8 09:22:13.000 2025 (UTC - 7:00)
System Uptime: 14 days 17:11:50.433
Process Uptime: 0 days 0:00:13.000
................................................................
............................
Loading unloaded module list
............
This dump file has an exception of interest stored in it.
The stored exception information can be accessed via .ecxr
(dfa8.628c): Access violation - code c0000005 (first/second chance not available)

+------------------------------------------------------------------------+
| This target supports Hardware-enforced Stack Protection. A HW based    |
| "Shadow Stack" may be available to assist in debugging and analysis.   |
| See aka.ms/userhsp for more info.                                      |
|                                                                        |
| dps @ssp                                                               |
|                                                                        |
+------------------------------------------------------------------------+

For analysis of this file, run !analyze -v
ntdll!NtWaitForMultipleObjects+0x14:
00007ffd`7c1df954 c3              ret
0:000> !analyze -v
*******************************************************************************
*                                                                             *
*                        Exception Analysis                                   *
*                                                                             *
*******************************************************************************

*** WARNING: Unable to verify checksum for superslicer.exe
[... 142 lines cut ...]

MODULE_NAME: Slic3r

IMAGE_NAME:  Slic3r.dll

STACK_COMMAND:  .ecxr ; kb ; ** Pseudo Context ** Pseudo ** Value: 4 ** ; kb

FAILURE_BUCKET_ID:  INVALID_POINTER_READ_c0000005_Slic3r.dll!Unknown

OS_VERSION:  10.0.26100.1

BUILDLAB_STR:  ge_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {37a38d71-3504-14e1-5b2f-fff8991f1972}

Followup:     MachineOwner
---------

````

Hopefully this crash dump file is useful.

jbeima, 2025-07-17:
I can confirm in the latest nightly build on Linux (GTK3) my STEP files open.