supermerill/SuperSlicer #4646: Crashing when trying to silce.

Trix44, 2025-07-12:
2.7 crashes instantly when i slices anything a error dose pop up but if flash's so fast i can not read it. i have tried a few different STL files something.  
 Here is the event log from windows event viewer.

Faulting application name: superslicer.exe, version: 2.7.61.6, time stamp: 0x686c3f48
Faulting module name: Slic3r.dll, version: 0.0.0.0, time stamp: 0x686c3f3e
Exception code: 0xc0000005
Fault offset: 0x0000000000b48321
Faulting process id: 0x0x3A20
Faulting application start time: 0x0x1DBF2E534E7FC61
Faulting application path: C:\SuperSlicer\SuperSlicer_2.7.61.6_win64_250707\superslicer.exe
Faulting module path: C:\SuperSlicer\SuperSlicer_2.7.61.6_win64_250707\Slic3r.dll
Report Id: a9c44a7d-64dc-48ac-8e40-4b270b82bbf2
Faulting package full name: 
Faulting package-relative application ID:

jbeima, 2025-07-12:
We will need the stls, and preferably the 3mf project file you are using to cause this.

If you have troubles uploading it, please zip it and then try again.

Trix44, 2025-07-13:
[stl3mf files.zip](https://github.com/user-attachments/files/21202616/stl3mf.files.zip)
Here are a few stl and 3mf files. They work fine in 2.5. let me know if you want more info.

jbeima, 2025-07-13:
Thank you. Will do.

jbeima, 2025-07-13:
Your project files do not contain any of your setting profiles. We need the print, filament, and printer settings you are truing to load these files into.

They open fine for m.

jbeima, 2025-07-13:
Does or did this happen in 2.7.61.5 for you?

Trix44, 2025-07-13:
[New 3mf.zip](https://github.com/user-attachments/files/21205859/New.3mf.zip)

Trix44, 2025-07-13:
so i installed 2.7.61.1-2.7.61.5 all worked, i went back and tried 2.7.61.6 and now it works???

jbeima, 2025-07-13:
@supermerill I have what we need on this one and will send it to you.

jbeima, 2025-07-13:
I have been able to re-create.

supermerill, 2025-07-17:
fixed in nightly