supermerill/SuperSlicer #4650: Thumbnails in RepRapFirmware stopped working between 2.5.60 and 2.7.61

Naomitr0n, 2025-07-16:
### What happened?

Without settings changes, thumbnails no longer correctly generate for the latest version of DuetWebControl, v3.6.0 (Duet 3 Mini 5+ WiFi, if it matters):
[image]

Here's the GCode files in question, from a project I'm still tidying up before posting to Printables. The V-slot one I have the project file for too:

[Broken GCode Thumbnails.zip](https://github.com/user-attachments/files/21245105/Broken.GCode.Thumbnails.zip)

I haven't changed my slicer settings for thumbnails in years, so it shouldn't be that I punched anything in wrong. It had been a hot minute since I updated my Duet (I was running an older version), so I updated both RepRapFirmware and DuetWebControl to latest to see if maybe there was a bug in DWC for some reason, but it didn't fix anything. Definitely something on SuperSlicer's side given the only change between thumbnails working and not working was changing which version of SuperSlicer I use. Looks like it stopped working between versions 2.5.60 and 2.7.61.2, as shown by the screenshot.

Here's the SuperSlicer Printer settings I've been using for years with DuetWebControl:
Thumbnail format: QOI (PanelDue requires QOI)
Small size: 48x48
Large size: 500x500

I didn't see any mention in the Releases page about changes to thumbnails, but something definitely got a bit wonky between the two.

### Project file & How to reproduce

Here's the three projects in the newer versions that aren't working correctly:
[Broken Projects.zip](https://github.com/user-attachments/files/21245153/Broken.Projects.zip)

I *believe* this one is the one that's working correctly and from an older version, but I'm not 100% sure.
[Plug Mount.zip](https://github.com/user-attachments/files/21245152/Plug.Mount.zip)

### Version

2.5.60 (works), 2.7.61.2 and beyond (doesn't work)

### Operating system

Win11

### Printer model

M3D Crane with Duet 3 Mini 5+ WiFi

jbeima, 2025-07-16:
Hey, I actually saw this as well, but it was a quote quote easy fix from a setting that needed to change. Leave it with me, since I run RepRap on a couple of my printers, so I will get you what you need to update.

jbeima, 2025-07-16:
Quick Question: What version of RRF are you running as well as which version of the DWC?

Naomitr0n, 2025-07-16:
It was an older one, but latest now, v3.6.0 for both firmware and DWC :)

jbeima, 2025-07-16:
Thanks, I will get back to you by Sat. Have a great night.

supermerill, 2025-07-18:
The qoi export didn't changed since its creation.
What are the difference in a (very small) thumbnail data between 2.5 and 2.7?

jbeima, 2025-07-18:
I do have a PanelDue on my machine, and I am fairly confident it displays png files while printing. However the menu for me, like going into the SD Card and clicking on a gcode file never displayed anything.

I will also confirm this, this weekend, since I am running some prints on the Modix.

Naomitr0n, 2025-07-18:
Looks like SuperSlicer 2.7 removed some options relating to the thumbnail that is causing all the ruckus.

> I think this maybe a bug in DWC not being able to auto-identify the image type and possibly defaulting to the wrong one.

Try these files out:
[Thumbnail Test.zip](https://github.com/user-attachments/files/21321873/Thumbnail.Test.zip)

I was able to unintentionally confirm this actually when running an A/B test case through. Made a new folder in DWC, uploaded the `2-5-60.gcode` file, thumbnail loaded as expected. Uploaded the `2-7-61-6.gcode` file, thumbnail also loaded which was _unexpected_. Original settings of 48x48 small, 500x500 large, QOI for both versions even. So I uploaded the `2-71-61-6.3mf` file, which showed that "asterisk" thumbnail thing that it does when it doesn't expect a thumbnail, and the `2-7-61-6.gcode` file suddenly _changed_ to a broken thumbnail.

I've tried messing around with this a bunch with the `2-5-60` versions and can't seem to get the GCode file to even _begin_ to try showing a broken thumbnail at all. So at least we know DWC is *capable* of recognizing the thumbnail, at least. That all said, I don't upload 3MF files at all, I was actually surprised I managed to get the thumbnail *working* to be honest with you.

Looking into the GCode files, I found something odd between them. I didn't change any of my settings, but the top of the GCode file changed from this:
```
; generated by SuperSlicer 2.5.60.0 on 2025-07-18 at 18:52:03 UTC

;
; thumbnail_QOI begin 48x48 1500
```

to this:
```
; generated by SuperSlicer 2.7.61.6 on 2025-07-18 at 18:53:46 UTC

;
; thumbnail begin 48x48 1500
```

`thumbnail_QOI` vs. `thumbnail`. So I looked at the settings screens under Printer Settings. 2.5.x:

[image]

2.7.x though:

[image]

@supermerill Is there some setting I'm missing that'll show those again or were they somehow removed? I can't even access them via search anymore, but it's definitely what's causing the issue.

So while DWC is definitely having a weird time auto-assuming thumbnail formats and getting it wrong, SuperSlicer seems to have removed my only ability to fix that in the 2.7.x branch, as well as whether or not I can show the bed in the thumbnail, and whether or not it prints at the start or at the end. Not sure why. Will update title again to reflect the actual issue now.

Naomitr0n, 2025-07-18:
Check this weirdness out. Upload `2-5-60.gcode` into an empty directory _first_. Then upload `2-7-61-6.gcode`. Should look like this:

[image]

Now upload `2-7-61-6.3mf`:

[image]

_boom_. Every time for me at least. This is only happening with the 2.7.x files because of the lack of ability to specify which thumbnail type. I cannot replicate this with the 2.5.x files that allow me to specify the thumbnail format of QOI.

Looks like a regression in SuperSlicer mixing with a weird bug in DuetWebControl. I don't know why DWC is deciding to refresh the thumbnail into a broken one when it gets it correct the first time, but I can't fathom that's correct. Will file a bug report in the DuetWebControl GitHub when I get a chance to write it up more properly.

jbeima, 2025-07-19:
Moving forward for RepRapFirmware this setting will be enabled by default.

supermerill, 2025-07-19:
[image]
maybe you have an old ui layout loaded? can you check in preferences->gui that your are in the standard layout?

[image]

Naomitr0n, 2025-07-19:
I haven't touched any layout stuff to be clear, whether it's in preferences or otherwise. I just download new versions and start using them. Is it possible there were changes between 2.5.x and 2.7.x in that area that might be relevant?

My settings look like this in 2.5.x:

[image]

And this in 2.7.x:

[image]

I haven't touched _any_ of this, so I'm not sure why version upgrades are being handled incorrectly, but this feels like something that should be fixed before the next "stable" release to me. Is there a settings migration/import from older version issue going on? Bare minimum it's weird that there doesn't seem to be any default chosen. I genuinely don't want to touch any of this and have it start working suddenly and then be unable to figure out what was going on.

> We did not remove any options in 2.7.61.X

@jbeima Then visibility of those options changed, however you want to approach the meat of the matter, which is why I named the bug report what I did. I haven't touched _anything_ that should hide them, meaning _this is indeed a bug with what is shown in the UI_. All due respect, this issue was named correctly, I don't know why your reply has such abrasive, standoff-ish language and why there was forcible removal of any mention of the exact issue at hand, but options vanishing due _only_ to downloading an updated version of SuperSlicer _is a bug_. Nothing is "fixed", and it is _not_ "working as intended".

2.5.x:
[image]

2.7.x:
[image]

I'm going to be reverting your change to the title of this issue as it completely sidesteps the problem at hand. Please leave the title intact until the _actual_ issue is fixed. As it stands, the title of this issue doesn't have anything to do with SuperSlicer, when it's fairly obvious there's changed behavior within SuperSlicer causing issues. I literally _only_ download new versions when they come out and start using them. Something changed without me doing anything to change that. _That is what a bug means_. Forgive me if I am sounding rude here, it does not appear that what I am writing is even being _read_ in the first place. Especially since right up top I wrote this:

> **_I haven't changed my slicer settings for thumbnails in years, so it shouldn't be that I punched anything in wrong._** It had been a hot minute since I updated my Duet (I was running an older [...]

I can't fathom how "working as intended" and _disabling dark mode to show options_  are even compatible thoughts. Let's slow down here instead of jumping to a lot of conclusions that aren't even accurate and don't even make sense either. @supermerill 's on the right track here, something's off. I don't think the end user should be expected to disable dark mode to get UI options to reappear after a version upgrade. That's unreasonable.

jbeima, 2025-07-20:
I would suggest you remove your config directory and start again with a fresh one.

You can test this by launching SuperSlicer with the --datadir parm.

./superslicer --datadir ./config-test

Once you have created the new directory you can copy your personal profiles over by copying the contents of:

/print/, /filament/, /printer/, and /vendor/.

After you have your new folder, DO NOT allow any older version to write to it.

jbeima, 2025-07-20:
This is now a new topic and unrelated to the original one. Which was the DWC not recognising the QOI contained in the gcode file, which is a bug in the DWC auto image recognition. Simply enabling the setting in include image type solves that. Hence our software works as expected. We are changing the default in our profiles moving forward to have this enabled by default. Hence updated for next version.

So the original intent of this issue is called correctly and the title has been changed back to what it was and should be.

supermerill, 2025-07-21:
@Naomitr0n 
Maybe the slicer did somethign wrong with the layout file in your config directory.
You can do as @jbeima said.
You can also just delete all .ui that are in your config directory, it should use the slicer's one.
This kind of issue can appear when different versions try to use the same configuration directory.
I'll try to change that for next version, to avoid these kind of issues.