supermerill/SuperSlicer #4651: Forced slice not working when hiding or unhiding an object from the list and the (Re)slice command is not working in 2.7.61.x

tehniemer, 2025-07-23:
### What happened?

I'm able to reproduce the issue by slicing a handful of objects and either hiding or unhiding one, which doesn't force a re-slice for some reason, and either using the command `Ctrl+R` or using the menu selection.

I do have objects off the build plate if that matters.

### Project file & How to reproduce

1.  Slice a plate
2.  Hide an object
3.  Forced re-slice is not triggered
4.  (Re)slice command does nothing

### Version

2.7.61.7

### Printer model

Voron Trident

jbeima, 2025-07-23:
We need your project file to look into this please.

tehniemer, 2025-07-23:
[A4T - WWBMG - Extruder Adapter \[xol-carriage\].zip](https://github.com/user-attachments/files/21388332/A4T.-.WWBMG.-.Extruder.Adapter.xol-carriage.zip)

Here it is.

tehniemer, 2025-07-27:
I kludged together a docker container based on some other projects. I'm by no means what I would consider competent, more trial and error approach with it. 

https://github.com/tehniemer/superslicer-vnc

jbeima, 2025-07-27:
This is why I asked which version you are running, since I can not recreate it. It works properly for me.

closed 2025-07-27 by jbeima

supermerill, 2025-09-01:
@jbeima 
> Just to be clear, it should not "auto" re-slice, it should simply present you with the Slice button to click when you are ready to slice again.

It should be able to auto-resclice if 'background processing' is selected.

@tehniemer
can't reproduce, can you zip your configuration directory and give it to me?
you can access it from  'help->show configuration directory'