supermerill/SuperSlicer #4652: Opening 'Mosaic from picture' causes segfault

netman74501, 2025-07-24:
### What happened?

Navigating to _Generate -> Mosaic from picture_ immediately causes a segmentation fault to occur. Not seeing this reported anywhere yet nor is it listed in the known issues.

Complete output from using GDB to obtain a backtrace:

```
$ gdb SuperSlicer-ubuntu_22.04-2.7.61.7.AppImage

GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from SuperSlicer-ubuntu_22.04-2.7.61.7.AppImage...
(No debugging symbols found in SuperSlicer-ubuntu_22.04-2.7.61.7.AppImage)

(gdb) run

Starting program: /home/vm-01/Downloads/SuperSlicer-ubuntu_22.04-2.7.61.7.AppImage 
[Detaching after fork from child process 1103680]
process 1103677 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 1103686]
process 1103677 is executing new program: /tmp/.mount_SuperSIFkIoL/bin/superslicer
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 1103690]
[New Thread 0x7ffff51ff640 (LWP 1103691)]
[New Thread 0x7ffff49fe640 (LWP 1103692)]
[New Thread 0x7fffeffff640 (LWP 1103693)]
[Thread 0x7fffeffff640 (LWP 1103693) exited]
[New Thread 0x7fffeffff640 (LWP 1103694)]
[New Thread 0x7fffeee45640 (LWP 1103695)]
[New Thread 0x7fffee644640 (LWP 1103696)]
[Thread 0x7fffee644640 (LWP 1103696) exited]
[Thread 0x7fffeee45640 (LWP 1103695) exited]
[New Thread 0x7fffeee45640 (LWP 1103697)]
[New Thread 0x7fffee644640 (LWP 1103698)]
[Thread 0x7fffeee45640 (LWP 1103697) exited]
[New Thread 0x7fffeee45640 (LWP 1103699)]
[New Thread 0x7fffede43640 (LWP 1103700)]
[Thread 0x7fffee644640 (LWP 1103698) exited]
[Thread 0x7fffeee45640 (LWP 1103699) exited]
[New Thread 0x7fffeee45640 (LWP 1103701)]
[New Thread 0x7fffee644640 (LWP 1103702)]
[Thread 0x7fffede43640 (LWP 1103700) exited]
[Thread 0x7fffeee45640 (LWP 1103701) exited]
[New Thread 0x7fffeee45640 (LWP 1103703)]
[New Thread 0x7fffede43640 (LWP 1103704)]
[Thread 0x7fffee644640 (LWP 1103702) exited]
[Thread 0x7fffede43640 (LWP 1103704) exited]
[Thread 0x7fffeee45640 (LWP 1103703) exited]
19:21:29: Debug: Adding duplicate image handler for 'JPEG file'
[New Thread 0x7fffeee45640 (LWP 1103708)]
[New Thread 0x7fffede43640 (LWP 1103709)]
[Thread 0x7fffeee45640 (LWP 1103708) exited]
[Thread 0x7fffede43640 (LWP 1103709) exited]
[New Thread 0x7fffede43640 (LWP 1103710)]
[... 220 lines cut ...]

A debugging session is active.

	Inferior 1 [process 1103677] will be killed.

Quit anyway? (y or n) y
```

### Project file & How to reproduce

Occurs on both Linux versions of the latest pre-release AppImage (2.7.61.7). To reproduce, run the AppImage and click _Generate -> Mosaic from picture_.

[project.zip](https://github.com/user-attachments/files/21397427/project.zip)

### Version

2.7.61.7

### Operating system

Ubuntu 22.04

### Printer model

Ender 3 S1

jbeima, 2025-07-26:
There is no images included in your zip and your project file simply has your printer profiles in it.

Please update an example image file you were using.

netman74501, 2025-07-26:
Project file is blank because I wasn't using one. The instructions on the upload section say to upload one even if you have no project because it contains the configuration.

As soon as I open the program, I click _Generate_ in the top menus then _Mosaic from picture_ and the program closes with a segfault. I don't have a chance to load any image.