supermerill/SuperSlicer #4653: Layer duration visualisation seems to be wrong

mwiercinski, 2025-07-24:
### What happened?

I am trying to test layer time sensitivity of a material by printing a cone, where extrusion speed is constant, cooling is off, any minimum layer time is set to zero. The only thing that should be affecting the layer time should the size of the cone slice, which is monotonically decreasing. 

The layer time preview shows layer times that are fluctuating! E.g. the bottom part of the cone prints in 3 seconds, then 2, 1, and again 2 seconds, and finally 1 second.  I think this may be an issue with rendering the preview, because the slicer reports constant linear and volumetric speeds throughout and no travel movements. As such the layer time should monotonically decreasing.

[image]

[image]

[image]

### Project file & How to reproduce

[Dragonburner cooling.zip](https://github.com/user-attachments/files/21418271/Dragonburner.cooling.zip)

### Version

2.7.61.1

### Operating system

Ubuntu 24.04.1 LTS

### Printer model

Custom toolhead based on dragonburner design, using Dragon ACE hotend.

jbeima, 2025-07-26:
I loaded your file, "hid" the part you have off the build surface, and then slicer the cone. Works as intended and can not reproduce what you shared above.

I suggest you either remove the off build plate part or hide it and try again.

mwiercinski, 2025-07-28:
I reproduced with a fresh build (`SuperSlicer-ubuntu_22.04-gtk2-2.7.61.7.AppImage`) and without the second object.

Does that depend on any dynamically linked dependencies?