supermerill/SuperSlicer #4657: Improving search navigation (remember position).

bukowa, 2025-07-27:
**Is your feature request related to a problem? Please describe.**
While searching for "first layer" settings I find it difficult to navigate into them ;
1. Click `ctr+f`.
2. Type `first layer`
3. Press keyboard `arrow down`.
4. Press keyboard `enter`.
5. Read where the setting is and what it changes (you still have to mouse over which isn't ideal).
6. Click `ctr+f` again to check another setting.
7. Position is not rememberd. You again have to mouse over settings to read their description.

**Describe the solution you'd like**
1. Remember position or ideally do not close the search window saving us from typing `ctr+f` again.
2. Show description for setting without needing to navigate into them with mouse over.

**Additional context**

[image]

jbeima, 2025-07-27:
This would be classed as a feature request. We have moved all Feature Request submissions to our website (https://www.superslicer.org/).

Please head over there and submit the Request A Feature Form.

bukowa, 2025-07-27:
> This would be classed as a feature request. We have moved all Feature Request submissions to our website (https://www.superslicer.org/).
> 
> Please head over there and submit the Request A Feature Form.

https://www.superslicer.org/sponsorship/request-and-sponsor-a-feature/

supermerill, 2025-09-04:
Thare is tooltip in the platter search window but not in the setting's one.
Just had to activate them, easy fix.