supermerill/SuperSlicer #4663: Incorrect work in extruder flow calibration (need change separator).

SKQR01, 2025-08-05:
### What happened?

 As far as I understand, the generated cubes should have the values of the extrusion multiplier in the range specified (from min to max, respectively), but I get completely strange values (for example, when min=0.9 and max=1.3, the values of the extrusion multiplier are distributed in the range 0-100, not 90-100) (see point 5).

But if I set values with comma, all will works corretly. I believe that the default value separator should be replaced with a comma.
 
[image]

[image]

[image]

### Project file & How to reproduce

1. Click on "Calibration"
2. Set Extruder Flow Calibration.
3. Go with default values.
4. Get incorrect result.
5. Change separators with comma.
6. Get correct result

### Version

2.7.61.7

### Operating system

Alt Linux

jbeima, 2025-08-05:
This is a support issue and not a bug... This needs to be submitted via Discord, and then you HAVE PATIENCE and WAIT for someone to get back to you.

SKQR01, 2025-08-06:
> This is a support issue and not a bug... This needs to be submitted via Discord, and then you HAVE PATIENCE and WAIT for someone to get back to you.

But the given values are generated by default (without my changes) and do not provide correct operation without changing them. Maybe it is worth fixing this?