supermerill/SuperSlicer #4667: Unable to add printer preset without material available

SarraceniaWilds, 2025-08-09:
### What happened?

When using the config wizard to select printer presets, if I choose any anycubic photon, there is no sla material available to select, and when I click 'finish' a popup alerts that no material has been selected, asks if you want to select material, and neither clicking yes nor no will allow you to proceed, because there are no materials available.

I wasnt sure if this was an issue from upstream but on the prusaslicer version i have I can select at least one material for this family of printers and proceed with it.

[image]
[image]

### Project file & How to reproduce

open the config window to add/remove printer presets
select an anycubic sla printer
unable to proceed

### Version

2.7.61.7

### Operating system

windows 11

### Printer model

anycubic photon

jbeima, 2025-08-09:
I am aware of challenges around Resin printers and profiles.

What I would suggest you try, is create a project in PS, save the 3mf project file, then open it in SS, and then simply save the 3 profiles. You can repeat this process over and over to migrate all of the additional profiles.

I will be up front though, I am not sure of the quality of slice that you will get out of SS for resin prints. Mainly because I am not aware of anyone that uses SS for resin printing at this point in time. :)

SarraceniaWilds, 2025-08-09:
> What I would suggest you try, is create a project in PS, save the 3mf project file, then open it in SS, and then simply save the 3 profiles. You can repeat this process over and over to migrate all of the additional profiles.

That doesnt work at all, just seems to import the object. No printer, material, or print settings make it over.