supermerill/SuperSlicer #4672: between-object-gcode is inserted at the wrong time.

TheAndr0id, 2025-08-24:
### What happened?

First off, this issue is upstream in PrusaSlicer, and I have entered an issue regarding it.  I don't normally use PrusaSlicer, so the fact that issue is gaining no traction is of little concern.  Here is a link to the issue on PrusaSlicer's Github:

https://github.com/prusa3d/PrusaSlicer/issues/14767

I'll keep things simpler here to avoid the same problems encountered over on PrusaSlicer's.

During sequential printing, between-object-gcode is being called after the hotend is moved to start the next object instead of when the previous object is completed.  The order being performed is:

1) Print Object 1.
2) Move print head to location of Object 2
3) Perform between-object-gcode
4) Lower head to layer 1 and start printing Object 2.

There are two problems.  First, there's no lift after Object 1.  Unless Z-Lift is enabled, the hot nozzle will drag through the last layer of Object 1 if Object 2 is opposite of where Object 1 completed.  This leaves a nice deep heat scar across the surface of the finished Object 1.

### Project file & How to reproduce

I've not attached a project file as configuration has no effect on this issue.  This is gcode generated in libslic3r.

1. Create two objects in the slicer (say Shape-Box)
2. Activate sequential printing, adjust spacing as needed for hotend spacing
3. Add a comment/code to between_objects_gcode
4. Slice and step through the gcode looking at travel and when the gcode in 3 is executed

### Version

2.5.59.13

### Operating system

Windows 7

### Printer model

Ender 3v2

TheAndr0id, 2025-08-24:
Since I've done some digging in PrusaSlicer, I'll point out what I _believe_ the change needed in SuperSlicer is.

In src/libslic3r/GCode.cpp in function GCodeGenerator:  

```
if (finished_objects > 0) {
                    _move_to_print_object(preamble_to_put_start_layer, print, finished_objects, initial_extruder_id);
                    std::string between_objects_gcode =
                        this->placeholder_parser_process("between_objects_gcode",
                                                         print.config().between_objects_gcode.value,
                                                         initial_extruder_id);
                    // Set first layer extruder temperatures, don't wait for it to reach the temperature.
                    // don't change the bed temp for each object, it can lead to auto-removel
                    this->_print_first_layer_extruder_temperatures(preamble_to_put_start_layer, print, between_objects_gcode, initial_extruder_id, false);
                    preamble_to_put_start_layer.append(between_objects_gcode).append("\n");
                } else
```
(I've confirmed that 2.5.59.13 and the current head version use the same code)

The `_move_to_print_object()` call should happen after the `this->placeholder_parser_process("between_object_gcode"... ` line.

I thought I'd quote part of one of my MANY messages in the PrusaSlicer issue which explains my reasoning:

> Think start-gcode/end-gcode. Start-gcode is called before the print head moves to the start location of a print right? End-gcode is called when the print is complete, but before finalization commands - for instance if M73 reporting is enabled. After my end-code there's a M73 P100 R0 telling Marlin that the print is complete.
> 
> Regardless if Z-lift isn't enabled the generated g-code is moving the hot nozzle directly across the freshly printed surface. In a single print this isn't a problem since end-gode can be used to lift the nozzle up and out of the way of the print. My assumption is that this is the point of between-object-gcode, kind of a combination end-start-gcode for between sequential objects.

I'm sorry if this is all a little short and blunt.  My experience over on PrusaSlicer's github placed a bit of strain on my patience.  I'm not looking for a overnight fix or anything, but I've just spent 2 weeks arguing over this issue to find out the other person isn't a PrusaSlicer developer....

supermerill, 2025-09-01:
I'll add a setting to set it before or after.
Also some new variables ot be used inside it.

supermerill, 2025-09-01:
implemented for 2.7.62

TheAndr0id, 2025-09-06:
Thank you!
(I'll leave this for you to close when you are done and happy with the commits)

closed 2025-09-06 by supermerill